cyoung/stratux #884: After upgrade to 029 US build, GPS no longer able to achieve fix reliably.

CharAznableLoNZ, 2024-03-20:
1. Stratux version: 1.6r1-eu029
2. Stratux config: Attached as startux_censored.conf.txt to get around github filters.  Removed wifi passwords and saved.

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: USB u-blox 8 GNSS (NMEA protocol)

    AHRS
      - [X] yes
      - [ ] no

    power source: 10Ah USB Battery Pack

    usb cable: Elbow Cable with Power Button

4. EFB app and version: (e.g., WingX Pro7 8.6.2)
FltPlan Go 5.0.17
    EFB platform: (e.g., iOS 9.2)
Android 14 AP1A.240305.019.A1
    EFB hardware: (e.g., iPad Mini 2)
Pixel 5a
5. Description of your issue:
After updating to the newest build, my stratux no longer can achieve a GPS fix.  I have tried rerouting the antennas to be sure they were not near the GPS unit.  Reimaging the microSD card with the stratux-v1.6r1-eu029-94438ef6-us.img to see if the OTA update failed.  The GPS will see satellites but will not achieve a fix reliably.  I have flown three times since updating and two of those times I needed to switch over to my phone GPS.  The rest of the ADS-B features of the Stratux work as intended. 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.
[stratux_censored.conf.txt](https://github.com/cyoung/stratux/files/14659068/stratux_censored.conf.txt)
[stratux.log](https://github.com/cyoung/stratux/files/14659069/stratux.log)

saabnut, 2024-03-20:
I may have the same problem, but ordered a new GPS as it was cheaper than
the time needed to troubleshoot. I have not tested it yet. FWIW, it seemed
to work fine for a while, then would occasionally drop out, then stopped
working altogether.

b3nn0, 2024-03-20:
The only real change regarding UBlox 8 in v029 was, that it now uses GPS+Galileo+Beidou instead of GPS+Galileo+GLONASS.

If you are able, you might want to try to compile Stratux yourself, and undo this commit for testing?
https://github.com/b3nn0/stratux/commit/3c872082649e51f72b368782987da38f5911105c
Maybe GLONASS works much better in the US than Beidou?

VirusPilot, 2024-03-20:
Can someone who has this issue provide a screenshot of the GPS page so that we can see what is happening?

CharAznableLoNZ, 2024-03-20:
I'm not sure if this is the page you're talking about but here is a screenshot of it tracking five satellites with no fix.   It does seem to drop off to one satellite often.  The unit is attached to the window of my house that has a great view of the western sky with nothing in the way beyond a tree here and there.  It has the same behavior in the plane as well.
While typing this I saw it surge up to eight satellites tracked then settle back down to four tracked.
[image]

N129BZ, 2024-03-21:
With GLONASS disabled and BeiDou enabled, I've never had a single resolved and fixed BeiDou satellite, and the best I've seen for the other satellites is maybe 5 or 6 in solution.  I disabled BeiDou and enabled GLONASS in gps.go and I get a minimum of 8 in solution.  Not sure if geographic location is pertinent, but I'm in central Texas
[image]

CharAznableLoNZ, 2024-03-21:
Mine has never achieved a fix inside.  This screenshot is the unit sitting outside on a table facing the same direction as the previous tests.  
It has failed 2/3 times to get a fix from inside the plane, the only thing between it and the outside world is a thin piece of plastic window.  I'm in the Seattle area.

[image]

N129BZ, 2024-03-22:
Mine is indoors and the ublox8 is in an upper USB receptacle above the 2 SDR's.  Im just curious, where are you (Virus Pilot) located?

VirusPilot, 2024-03-22:
I am located in the middle of Germany. Did you have significantly more satellites before migrating to v029? Would you never get more than ~10 satellites with v029? I did some measurements a few years ago and it turned out that with this particular setup of your GPS just above the SDRs you are loosing up to 10dB GPS signal. This is why I am recommending an external GPS in my shopping list: https://github.com/VirusPilot/stratux-pi4/wiki/Shopping-List-v3. Do you by chance have a USB extension cable so that you can move your GPYes away from the SDRs just for testing?

N129BZ, 2024-03-22:
I do have an external GPS, I will plug that in and let you know.  I have been purposely running in a "adversely difficult environment" just to see what absolute minimums can be while still giving me a good position fix.

N129BZ, 2024-03-22:
OK, this is with a NEO 8MU w/external antenna. It resolved 15 satellites almost immediately and signal strength is much better.  This is still inside my house, I expect it would be even better outside.  I'm going to recompile with BeiDou enabled and GLONASS disabled to see what that looks like... 
[image]

N129BZ, 2024-03-22:
And this is that exact same setup as mentioned, NEO M8U, nothing moved.  Recompiled with BeiDou enabled, GLONASS disabled.  A LOT different!! Going back to BeiDou disabled, GLONASS enabled. 
[image]

N129BZ, 2024-03-22:
And finally, returning to BeiDou disabled and GLONASS enabled... big difference.
[image]

N129BZ, 2024-03-22:
VirusPilot - I've never really had any issues with any of B3NNO's versions - but only noticed after this thread started that I was only getting around 8 resolved satellites with V29.  I think the change of disabling GLONASS may be not give the best results for the USA, or at least where I am located.  It might make sense for BeiDou or GLONASS to be configuration setting...

N129BZ, 2024-03-22:
VirusPilot - I've never had issues with any of B3NNO's versions - to me they have been nothing but positive for Stratux. I just happened to notice after this thread started that I was only getting around 8 resolved satellites with v29. I think the change of disabling GLONASS may be not give the best results for the USA, or at least where I am located. It might make sense for BeiDou or GLONASS to be a configuration option...

VirusPilot, 2024-03-22:
Alright, this is very helpful so we may revert back to GLONASS then.

VirusPilot, 2024-03-22:
If you by chance can do one more test, I would appreciate it. In gps.go the following change would be required:
GPS min 8, max 16 sats
GAL min 4, max 8 sats
BDS min 4, max 8 sats
It requires the following lines changed, starting with: https://github.com/VirusPilot/stratux/blob/de992daa057a1f7eb2793762493f066fb0bd5a14/main/gps.go#L517
to
galileo := []byte{0x02, 0x04, 0x08, 0x00, 0x01, 0x00, 0x01, 0x01}
beidou  := []byte{0x03, 0x04, 0x08, 0x00, 0x01, 0x00, 0x01, 0x01}

N129BZ, 2024-03-22:
Actually, I went ahead and added a conf setting for using GLONASS... of course, this was just for my own education  I'll go ahead and run with those changes suggested by VirusPilot... 
[image]

N129BZ, 2024-03-22:
I went ahead and added an option for using GLONASS, just for my own education. I will do the changes suggested by VirusPilot re-test, and let you know... 
[image]

VirusPilot, 2024-03-22:
Cool, feel free to submit a pull request accordingly.

N129BZ, 2024-03-22:
Here you go.  bytes adjusted as requested, running for 25 minutes
[image]

VirusPilot, 2024-03-22:
aha, now we are talking. Btw, if your GPS is a USB ublox M8 then I wonder why you only arrive at 2.1Hz update rate - our setting for the M8 is 10Hz like in my screenshot above.

N129BZ, 2024-03-22:
It's a ublox M8, however its a NEO M8U with on-board attitude sensors... Ive played with it a lot and it's possible I changed a setting in U-Center... Is that 10Hz assigned by a ConfigCommand, in Stratux, or is that supposed to be the M8 default?  If need be I can go into U-Center check.  
[image]

b3nn0, 2024-03-23:
I have now reverted to GLONASS instead of Beidou for the next version, as I was able to reproduce this more or less.
A web interface setting seems overkill to me. People tend to configure all kind of random stuff that they don't know about.. I have seen far too many completely miss-configured Stratuxes.
I'd rather find a default that works well for everybody, and I think in this case GLONASS is that default.

N129BZ, 2024-03-23:
As a further test, this looks much like the originally posted issue.  This is using an older GPYes/ublox7 usb, running so far for 1/2 hour. 
[image]

VirusPilot, 2024-03-24:
this is unrelated to any configuration, it is just poor GPS level

N129BZ, 2024-03-24:
Yes, after plugging the ublox7 into a usb extension cable and putting it on a window sill I'm getting a minimum of 10 resolutions + SBAS.