cyoung/stratux #887: http://slack.stratux.me/ doesn't connect

pcarew, 2024-08-26:
I was looking to join the slack group, but the webpage http://slack.stratux.me/ won't connect
Is there another way to join the slack discussion?

pp4life, 2024-08-26:
[stratuxadsb.slack.com](stratuxadsb.slack.com)

pcarew, 2024-08-26:
Thanks for responding, but stratuxadsb.slack.com won't let me sign in as I don't already have an account on that forum.
Response: It looks like there isn’t an account on Stratux ADS-B tied to this email address

pp4life, 2024-08-26:
Invite link I was able generate.

[https://join.slack.com/t/stratuxadsb/shared_invite/zt-2pe926hoj-~sehdIJ1JqZVz4VKb37chA](https://join.slack.com/t/stratuxadsb/shared_invite/zt-2pe926hoj-~sehdIJ1JqZVz4VKb37chA)

pcarew, 2024-08-27:
Thank you, that worked.

closed 2024-08-27 by pcarew