cyoung/stratux #888: RPI zero GPS Not communicating

AbbosMG, 2024-09-01:
Good evening, I have been using the StratuX on RPI 3b works awesome! However, recently been trying to make a more compact version on RPI 02w. 

The GPS unit when plugged to the RPI zero (through microUSB to USB-A adapter) the RPI won't communicate. I was wondering if someone could help me to figure out the issue. Attached are some images of the error as displayed and the GPS unit itself. [image][image][image][image]Thank you!

AbbosMG, 2024-09-02:
Log [image]

Velocipedic, 2024-11-10:
I'm getting the exact same error message (Pi Zero 2W) from an Adafruit Ultimate GPS USB module and a XA1110 qwiik connect GPIO model. Having looked through some of the documentation, the baud rate issue looks like it was resolved years ago. 

I'm willing to do some work to make them communicate, but I don't even know where to start.