cyoung/stratux #889: Need to control fan with AHRS and non stratux OS.

hmjgriffon1, 2024-09-15:
Hi, I reached out to stratux support and they directed me to the github and the slack that doesn't seem to exist anymore. I have a stratux box but have moved to another software. I still want to control the fan though, I am running the current raspberry pi OS and found this link online, would this still work with the AHRS 2.0?

https://www.reddit.com/r/stratux/comments/d97chy/ahrs_fan_controller/

is there a better way? Thanks!

pp4life, 2024-09-15:
Here's an invite link to the Slack. I don't know why slack.stratux.me doesn't work anymore. 
[https://join.slack.com/t/stratuxadsb/shared_invite/zt-2pe926hoj-~sehdIJ1JqZVz4VKb37chA](https://join.slack.com/t/stratuxadsb/shared_invite/zt-2pe926hoj-~sehdIJ1JqZVz4VKb37chA)

crewdogelectronics, 2024-09-20:
We switched to Discord:
https://discord.gg/ev2SwNSS